eufymake/eufyMake-PrusaSlicer-Release #1: Last used profiles not saved on re-launch

BradB111, 2023-11-15:
Hello,

On the release version (1.5.12) the profiles last used are not remembered when a new instance of AnkerMake Studio is opened.

e.g. If I select "AnkerMake M5 All-Metal Hotend 0.4 mm Nozzle" front the "printer" list. Close the application and then re-open AnkerMake Studio I should expect to see "AnkerMake M5 All-Metal Hotend 0.4 mm Nozzle" still selected but it **Always** defaults back to "AnkerMake M5 0.4 mm Nozzle" (The non all metal hot end version).

mgradwohl, 2023-12-05:
This one hit me today and ruined a print. If I have only added an M5C, then don't set the default to an M5.

AnkerKristy, 2024-02-04:
We apologize for this inconvenience have caused. This issue has been fixed in version V1.5.17, and we hope you can update to this version to try it out. If you have any further questions, please feel free to let us know. @BradB111

closed 2024-02-05 by AnkerAllen