eufymake/eufyMake-PrusaSlicer-Release #4: jansson.h missing from repo

mgradwohl, 2023-12-05:
Compiling on Windows, getting this

Error	C1083	Cannot open include file: 'jansson.h': No such file or directory	libslic3r_gui	D:\source\AnkerMake-PrusaSlicer-Release\AnkerStudio\src\slic3r\GUI\Network\DeviceObject.h	4

samuelwang1990, 2023-12-07:
Have you installed the jansson lib?

mgradwohl, 2023-12-09:
where in the instructions does it say to do that? 🙂

samuelwang1990, 2023-12-11:
in Readme.md you should compile the deps libs. 1. cd deps;mkdir deps_build;cd deps_build;
2. cmake  -DCMAKE_BUILD_TYPE=Debug -S .. -G "Visual Studio 16 2019";
3. Open the project and build the dependency library: {path}/AnkerStudio/deps/deps_build/destdir;
4. Set system environment variables:  THIRD_PART_ROOT = {path}/AnkerStudio/deps/deps_build/destdir;
5. mkdir AnkerStudio/build && AnkerStudio/build;
6. cmake -DCMAKE_BUILD_TYPE=Debug -S .. -G "Visual Studio 16 2019";
7. Open the project and build;

mgradwohl, 2023-12-12:
I did build the deps.  I'm pretty sure you left it out.

[image]

samuelwang1990, 2023-12-27:
Relevant instructions have been added, thank you for pointing it out.

joscha, 2024-01-16:
Closed via a26dc81cf9dfdd1e42037c92dea14f2307072d17

closed 2024-02-04 by AnkerAllen