eufymake/eufyMake-PrusaSlicer-Release #6: Update source code to latest

mgradwohl, 2023-12-23:
Current source code in this repo is out of date from the most recent release.

mgradwohl, 2024-01-04:
The latest release is 1.5.14, the code is 1.5.10. Please update.

mgradwohl, 2024-01-06:
The latest release is 1.5.15 the code here is 1.5.10 please update

joscha, 2024-01-16:
@aden.hu @pepper.hu ?

Anker-louisdu, 2024-02-04:
We apologize for this inconvenience have caused, We will update the code at the end of March. Thanks again for your attention on us. @mgradwohl @joscha

joscha, 2024-02-19:
@AnkerAllen this has been marked as closed, but currently HEAD is still at b99a3245b7f0e03f0b18479d09982cfca040fdc8 from about 2 months ago?

AnkerAllen, 2024-02-19:
@joscha Sorry, this should be a mistake, we will synchronize the code with the online version as soon as possible!

joscha, 2024-02-19:
Thank you @AnkerAllen - just to check, is this a one-way sync or if pull requests are made in this repository you'll be able to merge and incorporate them?

evilhamsterman, 2024-03-21:
@AnkerAllen I don't think you fully grasp what is being asked here. We are not just asking because we want to see the code but because your code is based on PrusaSlicer which is licensed AGPL v3 which means your code is Licenses AGPLv3 and you are currently in violation of section 6 of the license.

```
  6. Conveying Non-Source Forms.

  You may convey a covered work in object code form under the terms
of sections 4 and 5, provided that you also convey the
machine-readable Corresponding Source under the terms of this License
```

This is not just please and thank you, this is a legal requirement you are no obeying.

I'd also be careful you may be violating section 4 too

```
  You may convey verbatim copies of the Program's source code as you
receive it, in any medium, provided that you conspicuously and
appropriately publish on each copy an appropriate copyright notice;
keep intact all notices stating that this License and any
non-permissive terms added in accord with section 7 apply to the code;
keep intact all notices of the absence of any warranty; and give all
recipients a copy of this License along with the Program.
```

Your README mentions the code is licensed under the AGPLv3 but you do not include a copy of the license, at least not one that is easily findable. Traditionally you would include a LICENSE file in the root of your repository. 
The software itself includes a link to the LICENSE file in the PRUSA repo but that is not your code the license needs to be included with your code.

AnkerXuan, 2024-04-12:
> Current source code in this repo is out of date from the most recent release.

@mgradwohl Hello, the latest version 1.5.21 has been released, and the code has been updated to 1.5.21, including the license. Thanks for the correction.

vleeuwenmenno, 2025-09-10:
Meanwhile, we're at version 2.0.0.1 and the source code on this repo is WAY outdated.