eufymake/eufyMake-PrusaSlicer-Release #9: Using filenames with special characters fails the print

joscha, 2024-01-16:
When using long filenames for an STL file imported to a project, it seems the generated gcode (the location name is derived from it) fails to upload and print properly. I haven't exactly found out yet whether it's the length, special characters like commas and/or spaces in the filename that causes it.
I came here to research whether I could find and fix the issue but it seems to be blocked on updating the source code to match the actual latest version of the Ankermake Studio

AnkerAllen, 2024-02-04:
@joscha Thank you for your feedback, this is a known issue due to commas in filenamesthat we will address with a fix in the near future version, for now you can print by renaming the file  to remove commas.

joscha, 2024-02-04:
@AnkerAllen thank you, if you can point me towards the logic that handles this, I can attempt a fix and open a pull request?

AnkerTabWang, 2024-03-22:
@joscha We have fixed this issue in V1.5.19, where file names contain special characters such as commas or spaces.

closed 2024-03-22 by AnkerTabWang

joscha, 2024-03-22:
Great news, thank you! OOI, what commit was it fixed in, please?