eufymake/eufyMake-PrusaSlicer-Release #11: Cannot add multiple files via windows command prompt

xuteng0, 2024-02-02:
In the previous version an Ankermake, I was able to do the following:
1. have a list of 3D files ready to print
2. open the Windows command prompt, and enter the following to add each file to Ankermake slicer:
`"path\to\ankermake.exe" "path\to\3Dfile.stl"`

Sadly, the most recent update made this impossible. The steps above only work once when Ankermake slicer is not open. It can not add subsequent 3D files into the slicer. Many CAD software such as Fusion 360 rely on this method to directly send files to the slicer without the need to save an STL file. Please fix this issue in the next software update

Anker-louisdu, 2024-02-04:
Thank you for your feedback，I can not understand the first question, can you describe  it more detail.  the second question we will fix it in the near future version . @xuteng0

xuteng0, 2024-02-04:
Thank you for your prompt response. I appreciate your attention to my concerns. Let me provide additional details regarding my first question to ensure clarity:

In the previous version of Ankermake, I could utilize the Windows command prompt to add multiple 3D files to the slicer consecutively. The command looked like this:

`"path\to\ankermake.exe" "path\to\3Dfile1.stl"
"path\to\ankermake.exe" "path\to\3Dfile2.stl"
"path\to\ankermake.exe" "path\to\3Dfile3.stl"
`

However, with the most recent update, this method only works once when Ankermake slicer is not open. Subsequent attempts to add more files using the same command fail if the slicer is already running. In previous versions, this workflow was valuable for sending multiple files to the slicer without the need to save each STL file individually.

I hope this clarifies my concern. If there are any further questions or if you need additional information, please feel free to ask.

Thank you for your attention to this matter, and I look forward to the upcoming fixes in the near future version.

Anker-louisdu, 2024-02-04:
Thank you for your reply, we will fix this issue as soon as possible.

Anker-louisdu, 2024-03-11:
we have fix this issue @xuteng0

closed 2024-03-11 by Anker-louisdu

xuteng0, 2024-03-11:
Hi Louisdu

Thanks for fixing this bug. I have just upgraded Ankermake slicer and verified that it was indeed fixed

Best
Peter

Closed #11<https://github.com/ankermake/AnkerMake-PrusaSlicer-Release/issues/11> as completed.