eufymake/eufyMake-PrusaSlicer-Release #13: No way to modify filament settings for a single print

evilhamsterman, 2024-03-21:
It is possible to adjust print settings for a single print by selecting "Expert" mode, modifying the settings you wish and slicing. However other than selecting a filament type there is no way to modify filament settings just for a print, you have to edit the settings and save a new preset which if you are trying settings for a new filament can become tedious.

sweenbeen, 2024-05-02:
Yepp, agreed. This is a frustrating miss.

evilhamsterman, 2024-05-29:
This has been resolved, it is now possible to select the edit button and change filament settings for a single print.

It is annoying that when you try to change filament it discards all your print settings. It would be nice if there was a transfer option like with Orca Slicer.

closed 2024-05-29 by evilhamsterman