eufymake/eufyMake-PrusaSlicer-Release #14: Custom gcode always set to 25% speed

evilhamsterman, 2024-03-21:
When printing gcode generated by software other than AnkeMake Studio the print speed is always set to 25% unlike gcode sliced directly in AnkerMake Studio

yangboren1019, 2024-03-29:
> When printing gcode generated by software other than AnkeMake Studio the print speed is always set to 25% unlike gcode sliced directly in AnkerMake Studio

Please upload the gcode file to help us locate and analyze the problem.tks！

evilhamsterman, 2024-05-02:
It says it's 25% in the Android app, any gcode not generated by AnkerMake Studio. Here's an example 
[Right_PLA_Anker M5C.gcode.txt](https://github.com/ankermake/AnkerMake-PrusaSlicer-Release/files/15193595/Right_PLA_Anker.M5C.gcode.txt)