eufymake/eufyMake-PrusaSlicer-Release #18: Cannot open in windows 11

DavidKHG, 2024-04-27:
Since I download V1.5.21 released .exe file on windows 11, it installed well with admin option.
But no windows appear after run it (also no response with admin option)

The Lapto OS spec is here
- Edition: Windows 11 Enterprise
- Version: 23H2
- Installed date:	‎2024-‎04-‎25
- OS Buid: 22631.3447
- Experience: Windows Feature Experience Pack 1000.22688.1000.0

Here`s the log file when I tried to open it.
[AnkerMake_2024-04-27_13-23-25_3844.0.log](https://github.com/ankermake/AnkerMake-PrusaSlicer-Release/files/15137019/AnkerMake_2024-04-27_13-23-25_3844.0.log)

If  there is option to get legacy version .exe, plz share link or files

Best,
David Kim

AnkerAllen, 2024-04-28:
Hi @DavidKHG , thanks for the feedback!
If you need to install the software under the administrator account but use it under the non-administrator account, please try not to install the software to the Program Files directory of the system disk , there will be some user privilege issues that will cause the software not to start normally;

Please install the AnkerMake Studio software in a non-system disk directory, such as D drive or E drive;

DavidKHG, 2024-04-28:
Hi @AnkerAllen, 
Thanks for your idea, but currently I have single account on windows 11 only and also single C drive.
For previous version(maybe  V1.5.10), it gives no issue while using with Ankermake M5C in normal installation procedure.

I tried V1.5.21 to other labtop with windows 11 pro, but after install the ankermake studio, it gives same issue. No response after run the ankermake studio.exe

Is there any way to get pre-built V1.5.10 released version..?

Also curious about other users with windows 11 have same issue or not.

Best,
David

AnkerAllen, 2024-04-28:
@DavidKHG We have not received any feedback from other users at this time, this is most likely a system compatibility or permissions issue and the logs you provided did not turn up useful information. Can you tell us which path it is installed in if you can?
Alternatively you can download version 1.5.10 from [here](https://github.com/ankermake/AnkerMake-PrusaSlicer-Release/releases/download/v1.5.10/AnkerMake_Studio_V1.5.10_9.exe);

DavidKHG, 2024-04-28:
Hmm, Interesting...
When I installed V1.5.21 into "C:\Users\USER\AppData\Local\AnkerMake Studio" path, it did not work,
But if I re-install with V1.5.10_9 into same directory, it works well.

Also if I install V1.5.21 into "C:\Users\USER\Documents\ws\AnkerMake Studio" path, it also did not work,
but re-install with V1.5.10_9 into same directory, also works well.

So, I think the problem is not related with system authority issue..

I should have to work with legacy version first. Hope to be solved in a next release.
and if you need any additional info, let me know.

Thanks,

Best,
David

AnkerAllen, 2024-05-27:
Hi @DavidKHG,
Sorry the new version is a bit late. Now you can try our latest version [1.5.22](https://github.com/ankermake/AnkerMake-PrusaSlicer-Release/releases/tag/v1.5.22), which should solve the problem you mentioned!