eufymake/eufyMake-PrusaSlicer-Release #28: Slicer info box not visible following update to Anker Make Studio 1.5.2.3

boddysb, 2024-07-30:
[image]

boddysb, 2024-07-30:
to confirm, version detail 1.5.23.9 e1.8.0

ausminternet, 2024-08-01:
Same version here but on macOS. No Problems. Windows only?

Anker-louisdu, 2024-08-01:
We apologize for this inconvenience have caused,  We are investigating the issue, Thanks again for your attention on us. @boddysb

Anker-louisdu, 2024-08-02:
@boddysb you can changing the  computer's text scale to 125%  or 100%  to fix this issue.

Anker-louisdu, 2024-08-02:
system -> display -> scale and layout, then change the size of text 100%-150%

geonom, 2024-09-16:
Thank you, while that does kind of work - on High DPI displays, the text becomes incredibly small and hard to read. Hoping for a real fix.