eufymake/eufyMake-PrusaSlicer-Release #39: Log In Issues on AnkerMake Studio (The email or password you entered is incorrect.)

iamhmh, 2024-11-07:
Hi, I have a problem when I want to log in to the AnkerMake Studio software, here is :
-I click on Account > Log In (Nothing special so far);
-I enter my log in credentials and click on "Sign In";
-I select the country and press "Confirm";

And then, I got "The email or password you entered is incorrect.", but it's the correct credentials.

I click on "Forgot password?", I change my password, and I got the same message "The email or password you entered is incorrect.".

I try to create another account, I verify my other email, and then I got the same message "The email or password you entered is incorrect.".

I try to change the version of AnkerMake Studio, I took the 1.5.24, the 1.5.23, the 1.5.22 and the 1.5.19, I got the same issues in every version I've try.

Did anyone else have the same issue? Did someone fix it?

Thank you for your help !

Anker-louisdu, 2024-11-07:
Could you please provide your login account? @iamhmh @SantyagoSeaman

SantyagoSeaman, 2024-11-07:
@Anker-louisdu makeev.alex@gmail.com

spl1974, 2024-11-07:
Same here: spl1974@gmail.com

rosetown-kelly, 2024-11-07:
I am experiencing the same issue. I have tried resetting the password multiple times. I can log into the website portal not problem, but can not login to AnkerMake studio.

nfranke1, 2024-11-08:
Today Morning (Germany) it worked. Last Night no log in Possible for me as well.

spl1974, 2024-11-08:
Same here (Germany too). Login works again.

Anker-louisdu, 2024-11-08:
We have fixed the above issue, sorry for the inconvenience caused to you, thanks again for your attention and trust in our products, hope you have a good experience. @SantyagoSeaman @iamhmh @nfranke1 @rosetown-kelly @spl1974

iamhmh, 2024-11-08:
Same here (France), Login works again, Thanks for the support @Anker-louisdu !

closed 2024-11-08 by iamhmh

SantyagoSeaman, 2024-11-08:
@iamhmh Thank you so much!