eufymake/eufyMake-PrusaSlicer-Release #41: Invalid flow height supplied to new_from_config_width()

Patrick-ZV, 2024-12-03:
I just installed ankermake studio but when i load an stl file and slice it, i get the message on the screenshot.

Did i forget somthing?

[image]

MrGamy, 2025-02-14:
I have been having the same issue for months and multiple released versions. Support was not helpful, reinstalling and even purging %APPDATA% did not help either. Please let me know which logs or files you need to investigate this, I don't want to have to use different slicer software.

[image]

Imdagnir, 2025-03-03:
This is also an issue for me. I've installed it on two different devices and it appeared both times.